jsk-ros-pkg/jsk_aerial_robot #16: 8リンクのhydrusだとtransform controllerが落ちる

chibi314, 2017-04-06:
@tongtybj 
とりあえず8リンクのHydrus(まずは開リンク，閉リンク化はあとで考える）を作って
`roslaunch hydrus_transform_control hydrusx_bringup.launch simulation:=True headless:=False model:=octa_closed`
したら，transform controllerが落ちます．
printfデバッグで，おそらくhamiltonMatrixSolverの中で死んでいるということまでわかりました

chibi314, 2017-04-06:
さらにprintfデバッグをして，1250行目あたりの
`ces.compute(H);`
で死んでいることがわかりました．これはTHREE_AXIS_MODEの条件分岐の中で起こっています．
しかし，そもそも必ずTHREE AXIS MODEになるという現象が起きています．正八角形でも起こるので，何かがおかしいようです．

tongtybj, 2017-04-07:
@chibi314 
#19 で確認してみて、僕の手元のPCでは普通に動かせる。
問題は、今までは6リンクを上限に設計していたから、
以下のros messageは6に固定している:
- [aerial_robot_msgs::FourAxisCommand](https://github.com/tongtybj/aerial_robot/blob/devel/aerial_robot_msgs/msg/FourAxisCommand.msg)
- [aerial_robot_msgs::ControlTerm](https://github.com/tongtybj/aerial_robot/blob/devel/aerial_robot_msgs/msg/ControlTerm.msg)
- [aerial_robot_base::Pwms](https://github.com/tongtybj/aerial_robot/blob/devel/aerial_robot_base/msg/Pwms.msg) (これは8になっている)
- [hydrus_transform_control::RollPitchYawGain](https://github.com/tongtybj/aerial_robot/blob/devel/hydrus/hydrus_transform_control/msg/RollPitchYawGain.msg)

8リンク以上のものは作らないと考えているのなら、要素数を8に変更すればいいけど、安直だよね。

closed 2017-05-05 by tongtybj