jsk-ros-pkg/jsk_aerial_robot #36: Gazebo cheat mode

KahnShi, 2017-05-17:
@tongtybj 
1. Adjust torque_force rate 
2.  Increase joint velocity
3. Add yaw velocity control in /uav/uav topic (currently missing)

KahnShi, 2017-05-18:
@tongtybj 
Trying to use velocity control for joints but failed.
I did in following way
`rostopic pub /hydrus3/joints_ctrl sensor_msgs/JointState "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
name: ['']`
`position: [0]`
`velocity: [0, 0, -0.1]`
`effort: [0]" 
`
and simulation gave me such error:
[image]

tongtybj, 2017-05-18:
1. I did not implement the joint velocity control mode, only position control mode
2. The error means the number of ``` position[] ``` in  ``` sensor_msgs/JointState``` is not equal with the joint number(e.g. 3 for hydrus3). Since you only set the ```velocity[]``` with correct elements, but no elements in ```position[]```

tongtybj, 2017-05-18:
@KahnShi 
I think we should create another issue to discuss the joint velocity control mode

closed 2017-06-16 by KahnShi