jsk-ros-pkg/jsk_aerial_robot #445: jsk-ros-pkgへの移行

tongtybj, 2021-01-07:
案1: 全部jsk-ros-pkg/jsk_aerial_robotという新しいレポジトリに移す

安直

tongtybj, 2021-01-07:
案2: 飛行ロボットに特化した部分のみ、jsk_aerial_robotという新しいrepositoryをjsk-ros-pkgに以下に作る。

それ以外で、他のロボットにも使えそうな部分:
- [aerial_robot_control](https://github.com/JSKAerialRobot/aerial_robot/tree/master/aerial_robot_control) の最適制御の解法をjsk_controlにマージする
- [aerial_robot_estimation](https://github.com/JSKAerialRobot/aerial_robot/tree/master/aerial_robot_estimation) は自己状態推定で、他のロボットにも使えそう？ となると、jsk_robot_estimationというrepositoryを作る？また、[kalman_filter](https://github.com/JSKAerialRobot/kalman_filter)もjsk-ros-pkg/kalman_filterというrepositoryをつくる?

chibi314, 2021-01-07:
未発表のロボット，制御法などを実装すると発表前に公開されてしまう．

closed 2024-07-09 by tongtybj