jsk-ros-pkg/jsk_aerial_robot #459: STM32CubeIDEへの移行

chibi314, 2021-04-02:
- TrueSTUDIOの使用は非推奨になっており、CubeMXとIDEが統合されたSTM32CubeIDEへ移行したい。
- TrueSTUDIOのままでも良かったが、ubuntu 18.04にv9.3.0を入れるとツールバーが表示されないバグがあって使いづらい。

tongtybj, 2021-04-02:
同意

tongtybj, 2021-04-08:
@greenpepper123 

この件、ちょっと調べてみて、もしかんたんにできそうなら、取り組んでほしいです。
ポイントはCubeMXからのプロジェクト生成で、いままでの状態と競合しないことです。

tongtybj, 2021-04-08:
@greenpepper123 

で、テストはmater branchではなく、https://github.com/JSKAerialRobot/aerial_robot/pull/417　からスタートしたほうがいい

greenpepper123, 2021-04-13:
以下のようなことを行いました
1. JSKAerialRobot/aerial_robot:rtos 最新をpullして手元のneuronにTrueSTUDIOで書き込む
2. CubeIDEをインストール
3. "Import SW4STM32 or TrueSTUDIO project" で.projectがあるディレクトリ(aerial_robot_nerve/neuron/neuron_f4/TrueSTUDIO/neuron)を選択し、ポチポチする
4. エラー無しで成功したので、ビルドする。TrueSTUDIOの時と操作は同じ
5. ビルド成功したので、書き込む。stlink-serverが必要と言われるので、インストールする https://www.st.com/ja/development-tools/st-link-server.html
6. debや.sh経由でのインストールは謎にエラーで止まってしまい成功しなかった(18.04.5 on ThinkPad X495)が、バイナリを実行可能にして直接/usr/binに置いたら動くようになる
7. デバッグを行い、これまたTrueSTUDIOの時と同様に一度HAL_Init()のブレークポイントで止まるのでResumeし、終わったらTerminateする
8. neuronの青点滅が再開する。

chibi314, 2021-04-13:
cubeideで開ける状態をプルリクで送ってもらえますか？

greenpepper123, 2021-04-13:
CubeIDE単体で書き込もうとしたら従来と同じくデバッグを使うことになりますが、STM32CubeProg https://www.st.com/en/development-tools/stm32cubeprog.html を別で入れれば普通に書き込み出来ると思います。リリースビルドじゃないとFlash足りないとかにならない限りは不要そうですが。
その場合、stlink-serverのおかげで、USBデバイスが競合せずにCubeIDEとCubeProgの両方が同時に仮想的なST-Linkに接続したままに出来る(片方を一度disconnectする必要がない)と思われます。

greenpepper123, 2021-04-13:
neuronだけですが一応送ります
#461
#417とは違うトピックだと思ったので分けました

greenpepper123, 2021-04-13:
あと書き忘れたこととしてDebugはstlink-serverを入れた後もう一度何かエラーを吐きましたがRun->Debug Configurationsから適当に設定したら出来ました

chibi314, 2021-04-13:
書き込みは別にcubeideのデバッグでOKです．

greenpepper123, 2021-04-14:
spinalもやってみました
ros_lib周りの仕組みは一通り読んで理解したつもりなのですが、別にこちらもとりあえず変わらないかなと思ってneuronと同様にconvertしビルドしデバッグしてみました
特にエラーはなくLED的には正しく動くようになったのですが、serial.launchを立ち上げてみたところImuがbad checksumと言われました
移行のせいというより僕の環境またはもともとのコミットが悪い気がしているのでもう少し調べます
一応catkin clean && catkin buildは行いました
```
$ roslaunch spinal_ros_bridge serial.launch                                                                                          [45/2419]
... logging to /home/greenpepper/.ros/log/48b1871c-9d1e-11eb-922c-287fcf996afb/roslaunch-ubuntu-14928.log                                                                                          
Checking log directory for disk usage. This may take a while.                                                                                                                                      
Press Ctrl-C to interrupt                                                                                                                                                                          
Done checking log file disk usage. Usage is <1GB.                                                                                                                                                  
                                                                                                                                                                                                   
started roslaunch server http://ubuntu:42003/                                                                                                                                                      
                                                                                                                                                                                                   
SUMMARY                                                                                                                                                                                            
========                                                                                                                                                                                           
                                                                                                                                                                                                   
PARAMETERS                                                                                                                                                                                         
 * /rosdistro: melodic                                                                                                                                                                             
 * /rosserial_server/baud: 921600                                                                                                                                                                  
 * /rosserial_server/port: /dev/ttyUSB0                                                                                                                                                            
 * /rosversion: 1.14.10                                                                                                                                                                            
                                                                                                                                                                                                   
NODES                                                                                                                                                                                              
  /                                                                                                                                                                                                
    rosserial_message_info (rosserial_python/message_info_service.py)                                                                                                                              
    rosserial_server (spinal_ros_bridge/serial_node)                                                                                                                                               
    spinal_ros_bridge/rosservice_server_manager.py                                                                                                                                                 
                                                                                                                                                                                                   
auto-starting new master                                                                                                                                                                           
process[master]: started with pid [14963]                                                                                                                                                          
ROS_MASTER_URI=http://localhost:11311                                                                                                                                                              
[... cut ...]
[INFO] [1618403899.538081]: Setup service server on get_board_info [spinal/GetBoardInfo]
[ERROR] [1618403899.539261]: Creation of service server failed: Checksum does not match: 66fd709be8b736fcb084865c0e818d87,430f7d06257ff67465ab9e333831c45c
[INFO] [1618403899.541519]: Setup service server on set_board_config [spinal/SetBoardConfig]
[INFO] [1618403899.547838]: Setup service server on set_attitude_control [std_srvs/SetBool]
[ERROR] [1618403899.549134]: Creation of service server failed: Checksum does not match: a479adad61b9876d2f6bebef8341cf8b,c02825e9f2b2fd32e312b8e549b32163
```

greenpepper123, 2021-04-14:
別件ですが、rosserial_arduinoのように一発でcatkinから書き込みまで出来るようにならないかなあと思ったのですが、CubeIDEはその点に関してはTrueSTUDIOと特に変わらなそうです
https://ioloa.com/?p=296 この記事のようにMakefileは自前で書いてCubeIDEでデバッグだけ行うという手法もあるようですがやらないでしょう
ただ、TrueSTUDIOと違うのは一応CubeIDEはビルドするとDebugディレクトリにmakefileを生成しています
https://github.com/greenpepper123/aerial_robot/blob/rtos/aerial_robot_nerve/spinal/mcu_project/TrueSTUDIO/spinal/Debug/makefile
実際ビルドする際もこれのmake allを行っているだけのようなので、少なくとも依存関係が変わらない限りはmake allでビルドしてopenocdまたはst-flashで書き込めばコマンドラインから一発は実現できなくもなさそうです

tongtybj, 2021-04-14:
@greenpepper123 

> serial.launchを立ち上げてみたところImuがbad checksumと言われました

これはros_lib(spinal用)とros側(host PC)のmsg(ここではSpinal/Imu)のバージョンが違うからおきる問題と思う。
以下のコマンドで解決するはず
```
$ rosrun spinal make_libraries.py
```
これは、現在ros側の最新のmsg情報からros_libを生成するコマンドです。本当はCmakeListに`add_custom_target`としていれたほうがいいかもしれない。

tongtybj, 2021-04-14:
> 実際ビルドする際もこれのmake allを行っているだけのようなので、少なくとも依存関係が変わらない限りはmake allでビルドしてopenocdまたはst-flashで書き込めばコマンドラインから一発は実現できなくもなさそうです

お、これはいい！ 実際Deubgディレクトリでmake allしてみて、おなじ生成物が出たわけだね。
正直、ライトユーザはコマンドラインから使えるようにしたほうがいいかもね

greenpepper123, 2021-04-14:
> rosrun spinal make_libraries.py
もちろん一度したのですが、catkin buildしなおした後にもう一度するのを忘れていました、すみません。土曜辺りになってしまいそうですがやります。

greenpepper123, 2021-04-17:
やりなおしました。
とりあえず赤い方の`Creation of service server failed: Checksum does not match`は出なくなって、実際各topicも読めました(例えば/imu)
`[ WARN] [1618643202.456159431]: Rejecting message on topicId=2, length=87 with bad checksum.`は出ていて、1,2,3や151(spinal/Imu),158(servo/states)についても出たり出なかったりします。ただこれは1パケット失敗して出ているだけかなと勝手に思っています。

ただ、rqtが最初起動しなくて、Ctrl+Cを押すと
```
^C[ERROR] [1618643568.984192]: /imu_calib service call failed: transport error completing service call: receive_once[/imu_calib]: unexpected error [Errno 4] システムコール割り込み
^C[ERROR] [1618643570.895363]: /mag_declination service for get decalination call failed: transport error completing service call: receive_once[/mag_declination]: unexpected error [Errno 4] システムコール割り込み
```
これらがコンソールに出て、一応起動します
でIMU calibやBoard Configuratorなどで情報が確認できるのですが、試しにacc calibを押すとrqtがフリーズします(コンソールにはメッセージなし)
serial.launchの方には
```
[INFO] [1618643659.983523]: Setup service server on imu_calib [spinal/ImuCalib]
[INFO] [1618643659.986193]: Setup service server on mag_declination [spinal/MagDeclination]
```
がちゃんと出ていますし、rqtがフリーズしてもLEDの点滅も止まっていないしデバッグで動かしても別に何のハンドラも踏まないので、何の問題なのかわからないところです

このrtosブランチってどれくらい動作確認されていますか？
また、今はspinal+neuron1個または4個で連結してテストしているのですが、rqtとかも含めた動作確認をするならservoとか繋がないとエラーになりうるとかありますか？

greenpepper123, 2021-04-17:
普通にビルドの(すなわち本質的な)問題な気がしてきたのでもう少し自分で調べます
お待ちください

greenpepper123, 2021-04-17:
@chibi314 に見て頂いたんですが、
・CubeIDE上でspinal.iocがプロジェクトに含まれてすらおらず、その時点で理想的な移行が行われていない
・そのせいで、HALのバージョンが違うとか何かうまく行っていないのでは？
という話でした。
それから、origin/rtosでTrueSTUDIOで書き込み直してみたら上記のservice関連は問題なしだったので、やはり移行がうまく言っていないことがほぼ確実です。

それで、「spinal.iocだけCubeIDEに食わせてコード生成して、残りのユーザーコードをコピーしてきてインクルードパスとか設定し直せば確実ではないか」という話だったのですが、どう思いますか？ @tongtybj

greenpepper123, 2021-04-17:
あと、
[image]
iocを使おうとするとこのように出ますが、continueでいいですよね？

chibi314, 2021-04-17:
見てみるとバージョンが最新が1.16.1のところ，V1.16.0を使っていると言われているので，Migrateで最新にしてもいいかもしれません．
多分バグフィックスがあるだけだと思うので．

tongtybj, 2021-04-19:
> やはり移行がうまく言っていないことがほぼ確実です。
> それで、「spinal.iocだけCubeIDEに食わせてコード生成して、残りのユーザーコードをコピーしてきてインクルードパスとか設定し直せば確実ではないか」という話だったのですが、どう思いますか？

議論の場にいなかったので、完全に理解していないのですが・・・
移行がうまく行っていないというのは、以下のどれかな？
1) HAL周りのコードの生成がioc->CubeIDEの過程でうまく行ってない？ となると、「残りのユーザーコードをコピー、・・・」は意味ないのでは？ここちゃんと理解していない
2) ユーザーコードの部分の移行がうまく行っていない? なら、「残りのユーザーコードをコピー、・・・」は意味あると思っているけど

> 見てみるとバージョンが最新が1.16.1のところ，V1.16.0を使っていると言われているので，Migrateで最新にしてもいいかもしれません．

はい、パッチバージョンが変わっただけで、APIは変わっていないとおもう。

greenpepper123, 2021-04-19:
どちらかというと2でしょうか
そもそも公式の移行順序だとiocを使ってくれすらせず、TrueSTUDIOの.projectをCubeIDE用に少し変換してくれるだけで、そのせいでその変換がうまくいっていないのではないかということですね

chibi314, 2021-04-19:
@greenpepper123 

rtosブランチのプログラムをTrueSTUDIOで書き込んで動かしてみましたが，特に問題なさそうです．
やはり移行したことによって何か問題が起きているということになります．

tongtybj, 2021-04-19:
> そもそも公式の移行順序だとiocを使ってくれすらせず、TrueSTUDIOの.projectをCubeIDE用に少し変換してくれるだけで、そのせいでその変換がうまくいっていないのではないかということですね

公式の移行チュートリアルはioc (STM32CubeMX)を使っていないということ？

いずれにせよ、少し不可解な問題だと思うね。
> 試しにacc calibを押すとrqtがフリーズします(コンソールにはメッセージなし)

これは、
1) rosserialからデータが来なくなったときのフリーズなのか -> spinal, rosserial周りがおかしい
2) rqtが変な値を受け取ったからフリーズしたのか -> ホストPC側に問題
に切り分けられるけど、
@greenpepper123 はrqtがフリーズしたとき、他のトピックはちゃんとechoできたかな(/imuとか)。

あと、今使っているspinalって机の上に置いてあるもの？

greenpepper123, 2021-04-19:
>公式の移行チュートリアルはioc (STM32CubeMX)を使っていないということ？

そうですね。ほぼ情報量ないですが https://www.st.com/resource/en/user_manual/dm00613834-migration-guide-from-truestudio-to-stm32cubeide-stmicroelectronics.pdf これが公式です
実際、行われていることを見てもCubeMXのことは全然気にせずただTrueSTUDIOの.projectと.cprojectをCubeIDE用に変換するだけの機能に見えます。

>試しにacc calibを押すとrqtがフリーズします(コンソールにはメッセージなし)これは、
rosserialからデータが来なくなったときのフリーズなのか -> spinal, rosserial周りがおかしい
rqtが変な値を受け取ったからフリーズしたのか -> ホストPC側に問題
に切り分けられるけど、
@greenpepper123 はrqtがフリーズしたとき、他のトピックはちゃんとechoできたかな(/imuとか)。

できます。
実際何が起きているのかログを吐いてくれないので仕方なくデバッガで追跡したのですが、
rqtを開いた時Jsk_Lib/sensors/imu/imu_ros_cmd.cppのimuCalibCallback()までは呼ばれているのですがその後一生imuCalibCallback()が呼ばれ続けているように見えました
rosserialのserviceの仕様で繰り返し呼ばれるのかと思ったのですが、そもそも
https://github.com/greenpepper123/aerial_robot/blob/b740d12729bb6dc373c117add60f358f9436d9cf/aerial_robot_nerve/spinal/mcu_project/Jsk_Lib/sensors/imu/imu_ros_cmd.cpp#L62
ここのgetIMUCalibData()の後`res = imu_calib_res_;`にたどり着かず何故か急にimuCalibCallback()の先頭に戻るという挙動でした。
なので、フリーズするのは無限ループになっているからかなと思います。

>あと、今使っているspinalって机の上に置いてあるもの？
そうです。最初に頂いたspinalを全て使っています。
ただorigin/rtosで正常動作しているので個体の問題を疑っているのだとしたらそうではないと思います。

greenpepper123, 2021-04-19:
Project->Properties->C/C++ Build->Setting->MCU GCC Compiler->Optimization->Optimization levelを-O2から-O0へ
Project->Properties->C/C++ Build->Setting->MCU G++ Compiler->Optimization->Optimization levelを-O1から-O0へ
変更したら動きました。
https://github.com/greenpepper123/aerial_robot/commit/5150f43d5ced611346cade777c30261772ea0450
ちなみに両方-O1とか、gccがO0でg++がO2とかもダメで、両方O0に戻したらやはり動くことを確認しました。なので気のせいではないと思います。

なので、
・実機に載せてみてO0でも性能が足りるか確認
・結局spinal.iocが関連付けされていない状態のままなので、ちゃんとCubeMXでregenerateしていけるように再構成
をしていこうと思います。

tongtybj, 2021-04-19:
デバッグおつかれさまです。

思い返しみるとKeil ARM-MDKからTrueStudioに移行したときも、`-O`周りで躓いていました。いにしえのコードを引っ張りだして、grepしてみたら、`-O 3`になっていました。なので、TrueStudioでは`-O 3`は動かず、やむを得ず`-O 2`にした記憶があります。

今回の似たようなシチュエーションでしょうね。

tongtybj, 2021-04-19:
@greenpepper123 

あと、`-O0`にしてから、`rostopic echo /imu`がたまに途切れるように見えたけど、それは治ったの？

greenpepper123, 2021-04-20:
それは一時的なもの(おそらくデバッガ繋いでいたせい)で、再現していません。

greenpepper123, 2021-04-24:
spinalは-O0, neuronは-O2のままで、書き込んで試してみました
https://github.com/greenpepper123/aerial_robot/tree/hydrus_xi_under_actuated_model_planning2

普通に飛行することができました。spinalの負荷が高くなる条件というのが何か人為的に生み出せればまた試したいと思いますが、とりあえず
・spinalは-O0でとりあえずパフォーマンスに問題がなさそうなこと
・neuronは従来通り-O2で問題なく動きそうなこと
が検証できました

tongtybj, 2021-04-25:
僕も実験に付き合って、ちゃんと飛んでいるところを確認みました。
全機体のCubuIDEへの移行はこれでいけると考えられるが、

問題は, #461 は #416 をベースにしていて、
- rqtのboard configuratorからサーボエンコーダの設定を初回に行う手間がある
     - 外部磁気エンコーダの取り付けたは必須ではない、今まで通りdynamixel内部エンコーダは使える (詳細は#416)
- spinal::ServoInfo　が変わった -> aerial_robotのマイナーバージョンを一個上げる: v1.3.0

以上の手間とバージョン更新を許容できるのなら、近々 #416, #461 の順でマージしようと思うます。
@chibi314 @Takuzumi240 どうでしょうか

ちなみに、PrimeXを設置するためのMotive更新に伴い,  [mocap_optitrack (ros package)](https://github.com/JSKAerialRobot/aerial_robot/blob/master/aerial_robot_melodic.rosinstall#L19-L23)の変更も必要なので、
この件もちょうどv1.3.0への移行に合わせて一緒にやったほうがいいと思うね。