jsk-ros-pkg/jsk_aerial_robot #466: High load of mocap node

tongtybj, 2021-05-17:
@chibi314 

I have solved the problem of high load of mocap_node:
https://github.com/ros-drivers/mocap_optitrack/pull/62

I hope this PR would be merged. Before that, let's use [this patch branch](https://github.com/tongtybj/mocap_optitrack/tree/while_func_bug_debug).

tongtybj, 2021-05-18:
https://github.com/ros-drivers/mocap_optitrack/pull/62 is merged, and the developer has update the version to [0.1.3](https://github.com/ros-drivers/mocap_optitrack/commits/master)
We still need to wait for the debian package for several weeks, I guess.

Let's manually install `mocap_optitrack` from source for  the coming several weeks.

chibi314, 2021-05-19:
@tongtybj 
Should we close this PR?
https://github.com/JSKAerialRobot/aerial_robot_demo/pull/169

closed 2021-05-20 by chibi314