jsk-ros-pkg/jsk_aerial_robot #491: free(): double free detected in tcache 2

fanshi14, 2021-07-15:
Dear All,

I wonder if anyone has ever meet this error when launching Hydrus simulation?
> free(): double free detected in tcache 2

[image]

(I Googled, but could not find direct reason with ROS.)

System: Ubuntu 18.04 / ROS Melodic
Build succeeded without errors.
Tested branch: new master; 1.2.0 (both are same problem)

I attached the terminate output [here](https://drive.google.com/file/d/1vHzuS94BO9PNJmF8F_oKa7TKp80hatJd/view?usp=sharing).

Thanks in adavance!

tongtybj, 2021-07-15:
- Please use gdb to probe the error point:
   add `launch-prefix="gdb -ex run --args"` in https://github.com/JSKAerialRobot/aerial_robot/blob/master/robots/hydrus/launch/bringup.launch#L58

- Try to clean your old workspace and rebuild

fanshi14, 2021-07-15:
After using gdb and googled the output, I found the reason is because of gcc version.
When I reset gcc version and rebuild, it goes well.

Thank you very much @tongtybj !

closed 2021-07-15 by fanshi14