jsk-ros-pkg/jsk_aerial_robot #618: Discussion about new system structure

sugihara-16, 2024-06-20:
For developers,

In order to refine this aerial robot system into a more sophisticated OSS, I suggest the following reforms:
- Support ROS2
- Plugin-like firmware architecture
- Useful interface for light users

I would like to discuss the architecture of the system we should aim for and the specific tasks required to achieve it.

sugihara-16, 2024-06-20:
### Transition to ROS2
- Our system should support ROS2 for real-time and reliability reasons, while it should also support ROS1 for a while.
- In addition to the system structure, we also have to introduce new implementation to communicate with low level processor via ros because ROS2 doesn't support rosserial.
- Then, the specific tasks to achieve it are following:
  -  Complete separation of ros-dependent and -independent parts of the system.
  - Implement both ROS1 and ROS2version for ROS dependencies
  - Introducing [micro-ros](https://micro.ros.org/) instead of rosserial.
  
[image]

sugihara-16, 2024-06-20:
### Plugin structure for firmware
- To build firmware for a variety of hardware configurations, a plug-in system composed of appropriately modularized programs is required.
- Modularizing the program for each small function makes it easy to develop additional features.
[image]
- In addition, as software becomes a plug-in system, the entire system needs to be more clearly hierarchized.
[image]

### ToDo
- [ ] Implementation of a mechanism that enables setup of software configurations from higher-level processes
- [ ] Modularization of each feature

tongtybj, 2024-06-20:
I am just curious about what "Nervous system" stands for in the about figure.

tongtybj, 2024-06-20:
Thx for sharing the article. I also learnt the similar design philosophy from ros_control and gazebo:
- https://wiki.ros.org/ros_control
- https://classic.gazebosim.org/tutorials?tut=ros_control

sugihara-16, 2024-06-20:
Thx, these two are pretty useful examples.