jsk-ros-pkg/jsk_aerial_robot #628: [Mujoco] lack of ground truth mode for attitude estimation

tongtybj, 2024-10-27:
Mujoco does not support ground truth mode for attitude estimation like Gazebo:
https://github.com/jsk-ros-pkg/jsk_aerial_robot/blob/master/aerial_robot_simulation/src/simulation_attitude_controller.cpp#L63-L77

What is the reason to truncate this function @sugikazu75 ?

sugikazu75, 2024-10-27:
I think I could not get baselink velocity and angular velocity using MuJoCo' API. So I gave up to implement grond truth mode, which requires the true value of angular velocity.

tongtybj, 2024-10-27:
@sugikazu75 

I see. I also realized it after creating this issue. 
A simple workaround is to use the sensor plugin with zero noise to act as a ground truth generator.

- angular velocity: https://mujoco.readthedocs.io/en/stable/XMLreference.html#sensor-gyro
- linear velocity:  https://mujoco.readthedocs.io/en/stable/XMLreference.html#sensor-velocimeter

For linear velocity, the value is desribed in local frame. So frame conversion is necessary.