jsk-ros-pkg/jsk_aerial_robot #663: The role to  publish the joint model to spinal

tongtybj, 2025-01-17:
Currently, the joint model is published from `servo_bridge` to `spinal`: https://github.com/jsk-ros-pkg/jsk_aerial_robot/blob/master/aerial_robot_model/src/servo_bridge/servo_bridge.cpp#L399-L423.

But I think the should be better place to publish this topic simultaneously with `uav_info`.

sugihara-16, 2025-01-18:
Yeah, it's the more appropriate way.
I just wanted to utilize some existing code structures in `servo bridge` to treat joint parameters.

tongtybj, 2025-01-19:
Yeah, I do understand your purpose.
First of all, the mechanism and the place of publishing  `uav_model` can be still improved 
Let's refactor these functions in the near future.

sugihara-16, 2025-01-19:
OK!